plorefice/dht11-rs #4: ESP32 usage

shanemmattner, 2022-10-10:
I'm trying to use this library with the ESP32 and running into problems with the `gpio` declaration.  Any help would be appreciated.

Error messages on `cargo build`:
```
[shane@fedora dht11]$ cargo build
   Compiling dht11 v0.1.0 (/home/shane/Desktop/Rust/esp32_rust/dht11)
error[E0277]: the trait bound `Result<Gpio23<Input>, EspError>: _embedded_hal_digital_InputPin` is not satisfied
  --> src/main.rs:24:32
   |
24 |     let mut dht11 = Dht11::new(dht11_pin);
   |                     ---------- ^^^^^^^^^ the trait `_embedded_hal_digital_InputPin` is not implemented for `Result<Gpio23<Input>, EspError>`
   |                     |
   |                     required by a bound introduced by this call
   |
   = help: the trait `_embedded_hal_digital_InputPin` is implemented for `OldInputPin<T>`
   = note: required because of the requirements on the impl of `embedded_hal::digital::v2::InputPin` for `Result<Gpio23<Input>, EspError>`
note: required by a bound in `Dht11::<GPIO>::new`
  --> /home/shane/.cargo/registry/src/github.com-1ecc6299db9ec823/dht11-0.3.1/src/lib.rs:78:20
   |
78 |     GPIO: InputPin<Error = E> + OutputPin<Error = E>,
   |                    ^^^^^^^^^ required by this bound in `Dht11::<GPIO>::new`

error[E0277]: the trait bound `Result<Gpio23<Input>, EspError>: _embedded_hal_digital_OutputPin` is not satisfied
  --> src/main.rs:24:32
   |
24 |     let mut dht11 = Dht11::new(dht11_pin);
   |                     ---------- ^^^^^^^^^ the trait `_embedded_hal_digital_OutputPin` is not implemented for `Result<Gpio23<Input>, EspError>`
   |                     |
   |                     required by a bound introduced by this call
   |
   = help: the trait `_embedded_hal_digital_OutputPin` is implemented for `OldOutputPin<T>`
   = note: required because of the requirements on the impl of `embedded_hal::digital::v2::OutputPin` for `Result<Gpio23<Input>, EspError>`
note: required by a bound in `Dht11::<GPIO>::new`
  --> /home/shane/.cargo/registry/src/github.com-1ecc6299db9ec823/dht11-0.3.1/src/lib.rs:78:43
   |
78 |     GPIO: InputPin<Error = E> + OutputPin<Error = E>,
   |                                           ^^^^^^^^^ required by this bound in `Dht11::<GPIO>::new`

error[E0599]: the method `perform_measurement` exists for struct `Dht11<Result<Gpio23<Input>, EspError>>`, but its trait bounds were not satisfied
   --> src/main.rs:35:33
    |
35  |         let measurement = dht11.perform_measurement(&mut delay).unwrap();
    |                                 ^^^^^^^^^^^^^^^^^^^ method cannot be called on `Dht11<Result<Gpio23<Input>, EspError>>` due to unsatisfied trait bounds
    |
   ::: /home/shane/.rustup/toolchains/esp/lib/rustlib/src/rust/library/core/src/result.rs:504:1
    |
504 | pub enum Result<T, E> {
    | ---------------------
    | |
    | doesn't satisfy `<_ as embedded_hal::digital::v2::InputPin>::Error = _`
    | doesn't satisfy `<_ as embedded_hal::digital::v2::OutputPin>::Error = _`
    | doesn't satisfy `_: embedded_hal::digital::v2::InputPin`
    | doesn't satisfy `_: embedded_hal::digital::v2::OutputPin`
    |
    = note: the following trait bounds were not satisfied:
            `<Result<Gpio23<Input>, EspError> as embedded_hal::digital::v2::InputPin>::Error = _`
            `<Result<Gpio23<Input>, EspError> as embedded_hal::digital::v2::OutputPin>::Error = _`
            `Result<Gpio23<Input>, EspError>: embedded_hal::digital::v2::InputPin`
            `Result<Gpio23<Input>, EspError>: embedded_hal::digital::v2::OutputPin`

Some errors have detailed explanations: E0277, E0599.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `dht11` due to 3 previous errors
[shane@fedora dht11]$ 
```

`main.rs`
```rust
use std::thread;
use std::time::Duration;

// use embedded_hal::digital::v2::IoPin;
use embedded_hal::digital::v2::OutputPin;

use esp_idf_hal::delay;
// use esp_idf_hal::gpio;

use esp_idf_hal::peripherals::Peripherals;

use dht11::Dht11;

fn main() {
    esp_idf_sys::link_patches();
    sensible_env_logger::init!();

    let peripherals = Peripherals::take().unwrap();

    let mut led = peripherals.pins.gpio2.into_output().unwrap();

    let dht11_pin = peripherals.pins.gpio23.into_input();
    let mut dht11 = Dht11::new(dht11_pin);

    loop {
        println!("Toggle");
        led.set_high().unwrap();
        thread::sleep(Duration::from_millis(1000));

        led.set_low().unwrap();
        thread::sleep(Duration::from_millis(1000));

        let mut delay = delay::Ets;
        let measurement = dht11.perform_measurement(&mut delay).unwrap();
        println!("{:?}", measurement);
    }
}
```

`Cargo.toml`
```
[package]
name = "dht11"
version = "0.1.0"
authors = ["Shane <shanemmattner@gmail.com>"]
edition = "2021"
resolver = "2"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
embedded-hal = "0.2.7"
embedded-svc = "0.22.0"
esp-idf-hal = "0.38"
esp-idf-svc = "0.42.0"
esp-idf-sys = { version = "0.31.6", features = ["binstart"] }
log = "0.4"
sensible-env-logger = "0.2.0"
anyhow = { version = "1", features = ["backtrace"] }
dht11="0.3.1"

[build-dependencies]
embuild = "0.29"
anyhow = "1"
```

kolemikko, 2022-10-31:
Try putting dht11_pin into open_drain_output mode:
`let dht11_pin = peripherals.pins.gpio23.into_open_drain_output();`

shanemmattner, 2023-03-03:
Thank you @kolemikko , that was the correct solution.  Here is the working code with the latest esp-idf-hal crate:

```rust
use dht11::Dht11;
use esp_idf_hal::{
    delay::{Ets, FreeRtos},
    gpio::*,
    prelude::Peripherals,
};

fn main() {
    esp_idf_sys::link_patches();

    let peripherals = Peripherals::take().unwrap();

    let dht11_pin = PinDriver::input_output_od(peripherals.pins.gpio5.downgrade()).unwrap();

    let mut dht11 = Dht11::new(dht11_pin);

    loop {
        let mut dht11_delay = Ets;
        match dht11.perform_measurement(&mut dht11_delay) {
            Ok(measurement) => println!("{:?}", measurement),
            Err(e) => println!("{:?}", e),
        }
        FreeRtos::delay_ms(1000);
    }
}
```

`Cargo.toml`:
```rust
[package]
name = "dht11"
version = "0.1.0"
authors = ["Shane <shanemmattner@gmail.com>"]
edition = "2021"
resolver = "2"

[profile.release]
opt-level = "s"

[profile.dev]
debug = true # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[features]
pio = ["esp-idf-sys/pio"]

[dependencies]
esp-idf-hal = "0.40"
esp-idf-sys = { version = "0.32", features = ["binstart"] }
dht11 = "0.3.1"

[build-dependencies]
embuild = "0.30.4"
```

closed 2023-03-03 by shanemmattner